SNL-WaterPower/siweed #85: Add bill of materials to paper

ryancoe, 2021-05-26:
https://github.com/SNL-WaterPower/siweed/blob/919908d0400da31c2c2407996ca2bd9713debb87/documentation/siweedPaper.tex#L187-L208

nickross4444, 2021-06-15:
It will need to be converted to a PDF and uploaded before uploading the paper, but the content is finished in a spreadsheet.

ryancoe, 2021-06-15:
May be helpful: https://ctan.org/tex-archive/support/excel2latex?lang=en

ryancoe, 2021-06-22:
@nickross4444 - per our discussion today, please break out the BOM so that we have totals for:

1. All parts except computer and tank
2. Computer and tank
3. 1+2

Please also add a sentence to the paper along the lines of _Note that a SIWEED system can be assembled with any tank on hand (e.g., a fish tank) and any MS Windows PC running; as these components can contribute significantly to system cost, the BOM shown in Table XX lists the tank and computer separately from the remainder of the components._

nickross4444, 2021-07-22:
- [ ] Add beach decorations to BOM, in a separate section.

nickross4444, 2021-08-04:
- [ ] Add V2 order to BOM
- [ ] Format nicely.

nickross4444, 2021-10-05:
@Anwi505, could you update the links in the BOM to model numbers?

ryancoe, 2022-11-07:
@nickross4444 - I've updated the [`BOM.csv`](https://github.com/SNL-WaterPower/siweed/blob/master/documentation/BOM.csv) with [c1e42da](https://github.com/SNL-WaterPower/siweed/commit/c1e42da1eddf0b0589be438d6f492c54434de541). Can you check that the "V2 order" has been incorporated per [your checkbox above](https://github.com/SNL-WaterPower/siweed/issues/85#issuecomment-892846999); after that we may want to make a separate issue about putting a link in the paper, as there will be a tricky order of operations things there (new branch, release, add link to BOM, make Zenodo tag).

closed 2022-12-22 by nickross4444