SNL-WaterPower/siweed #90: Pop up boxes

nickross4444, 2021-06-18:
Graphic design isn't my strong suit, so please give me some feedback to make these boxes look better:
[image]
[image]

The text in the boxes can be modified easily by editing the text files in siweed\Processing\siWeedGUI\data. @ryancoe, if it is important that these are YAML files, let me know and I can try to make it work, but YAML doesn't seem to play well with java.

ryancoe, 2021-06-22:
@nickross4444 - per our discussion today, please:

- [ ] add pop-ups for all of the plots
- [ ] use a single text file to load the messages for these pop-ups

nickross4444, 2021-06-24:
- [ ] Improve the look of the boxes.

nickross4444, 2021-07-07:
I think the pop up boxes look fine now:

[image]

@ryancoe , could you check and modify the language in them, so we can close this issue?

nickross4444, 2021-07-07:
Also a note that the buttons are now bigger and color matching:
[image]

ryancoe, 2021-07-13:
@nickross4444 - what branch should I be looking at for this?

nickross4444, 2021-07-13:
In develop [here](https://github.com/SNL-WaterPower/siweed/blob/develop/Processing/siWeedGUI/data/popUpText.txt)

ryancoe, 2021-07-13:
Ok, I've made some minor updates with bd2a6388717cecadc7f27a517dba61e48776e347. I think we can close this for now.

closed 2021-07-13 by nickross4444