SNL-WaterPower/siweed #96: Doubt about the wave propagation mechanism of the Sandia Interactive Wave Energy Education Display (SIWEED) project

karenbrazil, 2021-07-07:
Good morning,
I'm Karen Soares, a student at the Federal University of Santa Catarina in Brazil, a Naval Engineering student and a member of the University's Laboratory of Acoustics and Vibrations.
We are currently working on a project to build a mini test tank for interaction with students during classes and researching the subject, I found the Sandia Interactive Wave Energy Education Display (SIWEED) project . And so i would like to know, if possible and relevant to you, what system did you use to cause the waves in the tank, ie, how did you perform this up and down movement, was a piston engine used? Could you provide me with this information or even the design of this system?

Thanks

ryancoe, 2021-07-07:
@karenbrazil - Thanks very much for your interest. We are working on a paper to give a more complete description of the system. In the meantime, you can reference the [system diagram](https://github.com/SNL-WaterPower/siweed/blob/d3d93abbe821c4af9602978c34950a012f8b12cc/documentation/diagrams/systemLayoutPNG.png) to see that we're using a stepper motor and ball-screw drive.

ryancoe, 2021-09-20:
@karenbrazil - I'll close this issue for now unless there's something more you need clarification on?

closed 2021-09-20 by ryancoe