SNL-WaterPower/siweed #97: Curriculum

nickross4444, 2021-07-15:
I could not find a simple way to make the curriculum documents diff trackable, so I've left them [here](https://github.com/SNL-WaterPower/siweed/tree/master/Curriculum) in docx and pptx form for now. If we really want to at some point, we can remake the documents in Latex. For now here is what needs to get done:

- [ ] Add to vocab list. @anwi505

ryancoe, 2021-07-16:
this works fine for now

ryancoe, 2021-07-20:
@Anwi505 - can you include some discussion about the scenarios in which a wave energy converter might be used? For example:

- Natural disaster
- Expensive alternatives (some  island communities rely diesel power)
- Others?

ryancoe, 2022-11-02:
With b7ec56930f4981199ebfbde88a69fa2f489e14ef, I moved this material into a single markdown document ([`educational_material.md`](https://github.com/SNL-WaterPower/siweed/blob/master/educational_material.md)) and distilled it a bit. Closing for now, although it would be great if this can be extended in the future.

closed 2022-11-02 by ryancoe