SNL-WaterPower/siweed #98: Power and scaling

ryancoe, 2021-07-20:
@Anwi505 - Can you please make some notes about the power/energy calculations you've done with the town.

Anwi505, 2021-08-02:
<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:w="urn:schemas-microsoft-com:office:word"
xmlns:m="http://schemas.microsoft.com/office/2004/12/omml"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Word.Document>
<meta name=Generator content="Microsoft Word 15">
<meta name=Originator content="Microsoft Word 15">
<link rel=File-List
href="file:///C:/Users/ajfomuk/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

<link rel=themeData
href="file:///C:/Users/ajfomuk/AppData/Local/Temp/msohtmlclip1/01/clip_themedata.thmx">
<link rel=colorSchemeMapping
href="file:///C:/Users/ajfomuk/AppData/Local/Temp/msohtmlclip1/01/clip_colorschememapping.xml">

<style>

</style>

</head>

<body lang=EN-US style='tab-interval:.5in;word-wrap:break-word'>

<p class=MsoNormal align=center style='text-align:center'><b><u><span
style='font-size:20.0pt;line-height:107%'>City Scale<o:p></o:p></span></u></b></p>

<p class=MsoNormal><b><span style='font-size:12.0pt;line-height:107%'>HO Model
Train Scale<o:p></o:p></span></b></p>

<p class=MsoListParagraphCxSpFirst style='margin-left:1.0in;mso-add-space:auto;
text-indent:-.25in;mso-list:l0 level2 lfo1'><![if !supportLists]><span
style='font-size:12.0pt;line-height:107%;font-family:"Courier New";mso-fareast-font-family:
"Courier New"'><span style='mso-list:Ignore'>o<span style='font:7.0pt "Times New Roman"'>&nbsp;&nbsp;
</span></span></span><![endif]><span style='font-size:12.0pt;line-height:107%'>7.2
in = 50 ft<o:p></o:p></span></p>

<p class=MsoListParagraphCxSpLast style='text-indent:-.25in;mso-list:l0 level1 lfo1'><![if !supportLists]><span
style='font-size:12.0pt;line-height:107%;font-family:Symbol;mso-fareast-font-family:
Symbol;mso-bidi-font-family:Symbol'><span style='mso-list:Ignore'>·<span
style='font:7.0pt "Times New Roman"'>&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; </span></span></span><![endif]><span
style='font-size:12.0pt;line-height:107%'>1 in = 6.944 ft<o:p></o:p></span></p>

<p class=MsoNormal><b><u><span style='font-size:14.0pt;line-height:107%'>Platform<o:p></o:p></span></u></b></p>

<p class=MsoListParagraphCxSpFirst style='text-indent:-.25in;mso-list:l2 level1 lfo2'><![if !supportLists]><span
style='font-size:12.0pt;line-height:107%;font-family:Symbol;mso-fareast-font-family:
Symbol;mso-bidi-font-family:Symbol'><span style='mso-list:Ignore'>·<span
style='font:7.0pt "Times New Roman"'>&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; </span></span></span><![endif]><span
style='font-size:12.0pt;line-height:107%'>Dimensions: 18 in x 13 in<o:p></o:p></span></p>

<p class=MsoListParagraphCxSpMiddle style='text-indent:-.25in;mso-list:l2 level1 lfo2'><![if !supportLists]><span
style='font-size:12.0pt;line-height:107%;font-family:Symbol;mso-fareast-font-family:
Symbol;mso-bidi-font-family:Symbol'><span style='mso-list:Ignore'>·<span
style='font:7.0pt "Times New Roman"'>&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; </span></span></span><![endif]><span
style='font-size:12.0pt;line-height:107%'>Scaled <b>dimensions:</b> 125 ft x
90.3 ft<o:p></o:p></span></p>

<p class=MsoListParagraphCxSpLast style='text-indent:-.25in;mso-list:l2 level1 lfo2'><![if !supportLists]><span
style='font-size:12.0pt;line-height:107%;font-family:Symbol;mso-fareast-font-family:
Symbol;mso-bidi-font-family:Symbol'><span style='mso-list:Ignore'>·<span
style='font:7.0pt "Times New Roman"'>&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; </span></span></span><![endif]><span
style='font-size:12.0pt;line-height:107%'>Scaled <b>area:</b> 11,287.5 ft^2<o:p></o:p></span></p>

<p class=MsoNormal><b><u><span style='font-size:14.0pt;line-height:107%'>WEC<o:p></o:p></span></u></b></p>

<p class=MsoListParagraphCxSpFirst style='text-indent:-.25in;mso-list:l1 level1 lfo3'><![if !supportLists]><span
style='font-size:12.0pt;line-height:107%;font-family:Symbol;mso-fareast-font-family:
[... 18 lines cut ...]
style='font-size:12.0pt;line-height:107%;font-family:Symbol;mso-fareast-font-family:
Symbol;mso-bidi-font-family:Symbol'><span style='mso-list:Ignore'>·<span
style='font:7.0pt "Times New Roman"'>&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; </span></span></span><![endif]><span
style='font-size:12.0pt;line-height:107%'>Scaled <b>area:</b> 837.7 ft^2<o:p></o:p></span></p>

<p class=MsoNormal><span style='font-size:12.0pt;line-height:107%'><o:p>&nbsp;</o:p></span></p>

<p class=MsoNormal><b><u><span style='font-size:14.0pt;line-height:107%'>Power
Usage<o:p></o:p></span></u></b></p>

Structure | Power Needed | Power Used
-- | -- | --
Lighthouse light | 1 kW bulb | ?
Lighthouse cottage | 20 kWh/day | 0.833
House | 30 kWh/day | 1.25
General Store | 14.3 kWh/day | 0.5958
Fishing store (beach) | 14.3 kWh/day | 0.5958
Storage building | 0.16 kW (10 16-watt tube bulbs) | ?
Shed | 0.04 kW (4 100-watt bulbs) | ?
Streetlights | 0.08 kW/light (5 lights) | ?
Total |   |  

</body>

</html>

ryancoe, 2021-08-03:
Great work @Anwi505.

- I would re-label the center column `Energy` (this is where you should put quantities with units of `kWh`)
- If the information you have is in units `kW`, then you are already at power and no additional calculation is necessary. Therefore, you can move the 1kW for the lighthouse light into the `Power Used` column.

Does this help?

Anwi505, 2021-08-03:
<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:w="urn:schemas-microsoft-com:office:word"
xmlns:m="http://schemas.microsoft.com/office/2004/12/omml"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Word.Document>
<meta name=Generator content="Microsoft Word 15">
<meta name=Originator content="Microsoft Word 15">
<link rel=File-List
href="file:///C:/Users/ajfomuk/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

<link rel=themeData
href="file:///C:/Users/ajfomuk/AppData/Local/Temp/msohtmlclip1/01/clip_themedata.thmx">
<link rel=colorSchemeMapping
href="file:///C:/Users/ajfomuk/AppData/Local/Temp/msohtmlclip1/01/clip_colorschememapping.xml">

<style>

</style>

</head>

<body lang=EN-US style='tab-interval:.5in;word-wrap:break-word'>

<p class=MsoNormal><b><u><span style='font-size:14.0pt;line-height:107%'>Power
Usage<o:p></o:p></span></u></b></p>

Structure | Daily Energy Use (kWh/day) | Average Power (kW)
-- | -- | --
Lighthouse lightbulb | 24 | 1.0
Lighthouse cottage | 20 | 0.833
House | 30 | 1.25
General Store | 14.3 | 0.5958
Fishing store (beach) | 14.3 | 0.5958
Storage building | 3.84 (10 16-watt tube   bulbs) | 0.16
Shed | 0.96 (4 100-watt bulbs) | 0.04
Lamp Posts (6) | 8.64 | 0.36
Total | 116.04 | 4.835

<p class=MsoNormal><span style='font-size:12.0pt;line-height:107%'><o:p>&nbsp;</o:p></span></p>

</body>

</html>

Anwi505, 2021-08-03:
Hello,

I fixed the power usage table. Please let me know if I need to edit anything else. Thank you.

-Anwi

nickross4444, 2021-08-05:
I've changed the power scaling in 617936e1254bf84e719bfafaa01bf5a945bd86f0 so that the meter goes to 5 kW(units unlabelled), and the 4 thresholds should also make practical sense(1kW, 2kW, 3kW, 4kW). Right now the meter has no labelled units. I'm not sure if we want to label it as kW, since the only other units in the GUI are mm and mNm(unlabelled), and kW would be scaled units, where mm are not.

nickross4444, 2021-08-18:
We've decided to leave the units on the power meter unlabeled.

ryancoe, 2021-11-15:
Please incorporate into paper @Anwi505

ryancoe, 2022-11-02:
Added to [educational_material.md](https://github.com/SNL-WaterPower/siweed/blob/master/educational_material.md), closing.

closed 2022-11-02 by ryancoe