SNL-WaterPower/siweed #99: Design Files in Paper

nickross4444, 2021-08-04:
These are the instructions for design files in the paper:

The complete design files must be either uploaded to an approved online repository, uploaded at the
time of submission on the online Elsevier submission interface as supplementary materials [CAD
files, videos,. . . ], or included in the body of the manuscript [e.g. figures]. The two approved online
repositories are Mendeley Data and the Open Science Framework [OSF instructions]. Mendeley
data: https://data.mendeley.com/ Open Science Framework: https://osf.io/ Open Science Frame-
work HardwareX instructions: https://osf.io/wgk7q/wiki/home/

- CAD files. Authors are encouraged to use free and open source software packages for creating
the files. For CAD files, OpenSCAD, FreeCAD, or Blender are encouraged, but if not available
source files from proprietary CAD packages such as Autocad or Solidworks and other drawing
packages are acceptable. OpenSCAD:
- 3D printing. Supplementary files that facilitate the digital replication of the devices are en-
couraged. For example, STL files for 3-D printing components. We recommend uploading
CAD files to the NIH 3D Print Exchange as Custom Labware and providing a link to the
location. NIH 3D Print Exchange: http://3dprint.nih.gov/
- Electronics. PCB layouts and other electronics design files can be uploaded to the Open
Hardware Repository or other repositories. Open Hardware Repository: http://www.ohwr.org/
- Software and firmware. All software files used in the design and operation of the hardware
should be included in the repository. Provide a description of software and firmware and use
extensive comments in the code.

My thought is we include the following files:
-Arduino sketches
-Processing sketches
-Tank CAD model. @Anwi505 is going to try adding the Plunger and WEC hardware to the current model. We might also be able to upload pictures as design files, but I'm not sure about that.
-Plunger CAD model (maybe included in tank CAD model)
-MiniWEC buoy dimensions or CAD(maybe included in tank CAD model)
[image]

I found these dimensions for an old buoy, but I'm not sure what the new values would be. @ryancoe, do you have any guidance for this? (new values, and would a diagram like above be better or a CAD model?)

On another note the guidance states that "All designs must be submitted under an open hardware license." I believe we've done this for the sketches, but I don't think we have for any of the CAD models. @ryancoe, how should I move forward with this, and what do I list in the paper when it asks to fill in the license?

Current CAD model:
[image]

ryancoe, 2021-08-09:
A couple of thoughts/responses:

- Repository for uploading: this will be our Github repository, it already contains many of the required items (e.g., Arduino source, Processing source, etc.)
- CAD files
  - Plunger: just sent an email to @nickross4444 with an old archive that I think contains this
  - WEC: Please measure, but a I believe the current WEC has a major diameter of 16 cm: 
[miniWaveTank.xlsx](https://github.com/SNL-WaterPower/siweed/files/6955253/miniWaveTank.xlsx), @nickross4444, can you work with @Anwi505 to get a CAD model made?
- License: we are already using a [GPL license](https://github.com/SNL-WaterPower/siweed/blob/master/COPYING) that I think will satisfy this requirement

Did I cover everything?

nickross4444, 2021-08-17:
@ryancoe

ryancoe, 2021-08-18:
Thanks for digging into this @nickross4444. Please take a look at this page about integrating Zenodo w. Github. I haven't done this before, but others in our group have (maybe @ssolson?). If I understand it correctly, this will create a [DOI](https://en.wikipedia.org/wiki/Digital_object_identifier
) pointing to specific commit in our repo.

https://guides.github.com/activities/citable-code/

nickross4444, 2021-09-23:
Mostly finished in 663d029da2039ec0f449b3e0388ad48e68e9e34b. Decided to reference "online repository" instead of linking to each file, as other HardwareX papers have done. Now just waiting for the CAD model.

As a note, the course of action for finalizing the link to the repo in the paper will be as follows:

- when files are ready, create a new branch and delete the paper and other unnecessary files from that branch.
- Create a release out of that clean branch and upload the release to Zenodo.
- Link to the Zenodo repo in the paper in Master.

ryancoe, 2021-11-15:
@nickross4444 - please give status update here

nickross4444, 2021-11-17:
Once issues #3, #85, #102 are complete, we can create the new branch that will get uploaded to Zenodo and finalize the paper from there.

ryancoe, 2022-11-02:
Minor detail, but I think the key step with Zenodo is to tag a release (https://docs.github.com/en/repositories/archiving-a-github-repository/referencing-and-citing-content), not make a specific branch. Are you looking at something different?

nickross4444, 2022-11-04:
As detailed [here](https://github.com/SNL-WaterPower/siweed/issues/99#issuecomment-926076653), we'll need to create a release branch first, not only for gitflow but also to get rid of the paper itself and any other redundant documents from the release.

nickross4444, 2022-12-23:
@ryancoe I've created the branch for the release and the release, but in order to get it on zenodo with git integration, whoever owns the repo needs to log into zenodo through their github account. I think you should be able to do it in a couple minutes, but I am unable to.

ryancoe, 2023-01-04:
Ok, I think I did it: https://zenodo.org/record/7502670. Please close this issue if you agree.

nickross4444, 2023-01-24:
@ryancoe I think it's missing the CAD files and the educational material. Besides that good to go.

ryancoe, 2023-02-14:
This had me scratching my head for a bit, but the files _**are**_ actually there. The Zenodo website preview does not include all files. 

[image]

When I download the `.zip` file from Zenodo, I do indeed see the CAD files.

<details>
  <summary>File listing</summary>

```
total 64
drwxrwxr-x@ 5 rcoe  SANDIA\Domain Users   160B Dec 22 16:40 Arduino
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   3.1K Dec 22 16:40 CODE_OF_CONDUCT.md
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users    34K Dec 22 16:40 COPYING
drwxrwxr-x@ 4 rcoe  SANDIA\Domain Users   128B Dec 22 16:40 Processing
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   5.6K Dec 22 16:40 README.md
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   1.5K Dec 22 16:40 contributing.md
drwxrwxr-x@ 6 rcoe  SANDIA\Domain Users   192B Dec 22 16:40 documentation
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   8.5K Dec 22 16:40 educational_material.md

./Arduino:
total 0
drwxrwxr-x@ 5 rcoe  SANDIA\Domain Users   160B Dec 22 16:40 WEC_Controller
drwxrwxr-x@ 5 rcoe  SANDIA\Domain Users   160B Dec 22 16:40 WaveMaker_Controller
drwxrwxr-x@ 7 rcoe  SANDIA\Domain Users   224B Dec 22 16:40 libraries

./Arduino/WEC_Controller:
total 20
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   3.5K Dec 22 16:40 Interrupts.ino
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   2.2K Dec 22 16:40 Serial.ino
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   8.0K Dec 22 16:40 WEC_Controller.ino

./Arduino/WaveMaker_Controller:
total 20
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   4.8K Dec 22 16:40 Interrupts.ino
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   2.3K Dec 22 16:40 Serial.ino
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   7.8K Dec 22 16:40 WaveMaker_Controller.ino

./Arduino/libraries:
total 4
drwxrwxr-x@ 12 rcoe  SANDIA\Domain Users   384B Dec 22 16:40 DueTimer
drwxrwxr-x@  5 rcoe  SANDIA\Domain Users   160B Dec 22 16:40 Jonswap
drwxrwxr-x@ 13 rcoe  SANDIA\Domain Users   416B Dec 22 16:40 MD_AD9833
drwxrwxr-x@ 10 rcoe  SANDIA\Domain Users   320B Dec 22 16:40 SuperDroidEncoderBuffer
-rw-rw-r--@  1 rcoe  SANDIA\Domain Users    87B Dec 22 16:40 readme.txt

./Arduino/libraries/DueTimer:
total 40
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   7.0K Dec 22 16:40 DueTimer.cpp
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   2.5K Dec 22 16:40 DueTimer.h
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   1.1K Dec 22 16:40 LICENSE.txt
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   3.5K Dec 22 16:40 README.md
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   5.2K Dec 22 16:40 TimerCounter.md
drwxrwxr-x@ 5 rcoe  SANDIA\Domain Users   160B Dec 22 16:40 examples
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   324B Dec 22 16:40 keywords.txt
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   294B Dec 22 16:40 library.json
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   391B Dec 22 16:40 library.properties

./Arduino/libraries/DueTimer/examples:
total 0
drwxrwxr-x@ 3 rcoe  SANDIA\Domain Users    96B Dec 22 16:40 AvaliableTimer
drwxrwxr-x@ 3 rcoe  SANDIA\Domain Users    96B Dec 22 16:40 MultipleTimers
drwxrwxr-x@ 3 rcoe  SANDIA\Domain Users    96B Dec 22 16:40 SimpleTimer

./Arduino/libraries/DueTimer/examples/AvaliableTimer:
total 4
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   1.5K Dec 22 16:40 AvaliableTimer.pde

./Arduino/libraries/DueTimer/examples/MultipleTimers:
total 4
[... 1763 lines cut ...]
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   490K Dec 22 16:40 U Bracket.SLDPRT
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   151K Dec 22 16:40 double S Top Rods -  - Copy.SLDPRT
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   125K Dec 22 16:40 mini bracket.SLDPRT
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   140K Dec 22 16:40 rod connector.SLDPRT
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   151K Dec 22 16:40 short Top Rods - .SLDPRT
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   146K Dec 22 16:40 wave brace.SLDPRT

./documentation/diagrams:
total 6984
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users    21K Dec 22 16:40 Control Algorithm for PID Controller.png
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   224K Dec 22 16:40 DUE pin layout.png
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   562K Dec 22 16:40 SIWEED_CAD.png
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   4.4M Dec 22 16:40 Sandia_Wave_Tank_Exhibit_10x10_inlinebooth.png
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   145K Dec 22 16:40 SystemLayout.pdf
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   232K Dec 22 16:40 SystemLayout.png
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users    99K Dec 22 16:40 SystemLayout.vsdx
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   130K Dec 22 16:40 TorqueConstant.pdf
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users    91K Dec 22 16:40 TorqueError.pdf
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users    47K Dec 22 16:40 WaveMakerFlowChart.vsdx
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   379K Dec 22 16:40 mega pin layout.png
-rw-rw-r--@ 1 rcoe  SANDIA\Domain Users   489K Dec 22 16:40 siweed_guiScreenShot.png

```

</details>

nickross4444, 2023-02-14:
Okay, interesting. I'll get everything else wrapped up soon then.

commit 3176ddf6 (2023-03-20): finished all sections of paper. Closes #99 Closes #102
closed 2023-03-20 by nickross4444