SNL-WaterPower/siweed #101: wave maker tuning

ryancoe, 2021-08-31:
After completing #3 and #100...

1. Run a test matrix of regular waves with a range of frequencies and amplitudes (e.g., f = [1, 1.5, 2, 2.5, 3, 3.5, 4]Hz, A = [5, 10, 20, 30, 40]mm)
2. Compare amplitude measured by wave probes against the desire amplitude
3. Present results here (we'll decide a course of action at that point)

We will present this work in Section 7 of the HardwareX paper.

ryancoe, 2022-11-02:
Note that we had originally conceptualized this issues a wave maker **_tuning_** activity. We will limit this issue to a first step of **_characterization_** and leave **_tuning_** to a separate issue.

nickross4444, 2023-02-18:
I'm not 100% sure this is worth tackling. Not only did we observe significant capacitive coupling when touching the probes, but the data looks really noisy inconsistent to me:
[image]
(position determined by encoder)
plotted with:
[wavemakercharacterize.txt](https://github.com/SNL-WaterPower/siweed/files/10772765/wavemakercharacterize.txt)

ryancoe, 2023-02-20:
You may be right @nickross4444. Could you upload a zip of the data for me to look at?

nickross4444, 2023-03-16:
Data used for the previous plots: 
[P steps.csv](https://github.com/SNL-WaterPower/siweed/files/10995433/P.steps.csv)
[D steps.csv](https://github.com/SNL-WaterPower/siweed/files/10995435/D.steps.csv)

nickross4444, 2023-03-16:
Sorry about that, they should be correct now.