SNL-WaterPower/siweed #113: Paper Proofreading

nickross4444, 2023-03-20:
The paper has reached a completed state. In this issue, could we(@Anwi505 @DeepFriedDerp @ryancoe) post our revision suggestions before submitting?

[siweedPaper.pdf](https://github.com/SNL-WaterPower/siweed/files/11013383/siweedPaper.pdf)

ryancoe, 2023-03-20:
@nickross4444 - Great stuff. Please see suggested changes here: [siweedPaper.pdf](https://github.com/SNL-WaterPower/siweed/files/11023240/siweedPaper.pdf). I'd like to do one more round of reviews before submission.

@DeepFriedDerp @Anwi505 @delaneyheileman @stespenc83 @gbacelli - Please add any additional comments by 3/24 COB.

stespenc83, 2023-03-28:
@nickross4444 Sorry for not getting around to commenting on time. If it's not too late, I would say it looks like it could use a summarizing paragraph at the end. It felt like it ended very abruptly. Maybe recap the primary achievements, major lessons learned, and potential future work.

nickross4444, 2023-05-01:
My sincere apologies for the delay. Attached and pushed is the revised paper.
[siweedPaper.pdf](https://github.com/SNL-WaterPower/siweed/files/11365068/siweedPaper.pdf)